HITSZ-NRSL/Dynamic-VINS #1: Failed to load nodelet

GuoPingPan, 2022-08-11:
[image]

你好，我在运行时遇到了以上问题，单独运行yolo_ros正常。我已按照安装目录完成了ceres和Sophus库版本的修改，catkin_make一路绿灯。请问是什么问题呢？
[image]

GuoPingPan, 2022-08-11:
且希望获得 `OpenLORIS-cafe` 云盘的下载路径

jianhengLiu, 2022-08-12:
1. 有可能是由于使用了非ros版本的opencv
可以参考这个issue：https://github.com/jianhengLiu/VINS-RGBD-FAST/issues/3

2. OpenLORIS-cafe的下载路径可以打开[OpenLORIS](https://shimo.im/docs/HhJj6XHYhdRQ6jjk/read)里面有百度网盘的下载链接

GuoPingPan, 2022-08-12:
非常感谢您的答复，我通过指定OpenCV版本解决了该问题。

closed 2022-08-12 by GuoPingPan

kukulinxin, 2023-11-28:
> 非常感谢您的答复，我通过指定OpenCV版本解决了该问题。

你好，我想询问一下，这里指定Opencv版本是指安装ros时自带的opencv吗？还说版本对上就好了，比如ros melodic 一开始的opencv版本时3.2.0