HITSZ-NRSL/Dynamic-VINS #2: No Image received

BloodLemonS, 2022-08-14:
您好，当我按照文档运行的roslaunch的时候没有feature_image和sematic_image,请问是哪里出了问题吗？？
[image]

BloodLemonS, 2022-08-14:
另外他在运行的时候报IMU信息混乱
[image]

jianhengLiu, 2022-08-14:
1. 修改配置文件`Dynamic-VINS/blob/main/config/openloris/openloris_vio.yaml`中的
```
show_track: 0           # publish tracking image as topic
```
为
```
show_track: 1           # publish tracking image as topic
```
即可获得可视化输出

2. 出现imu warning一般不影响结果，但若持续长时间的出现会影响性能。

BloodLemonS, 2022-08-15:
已经解决了，感谢

closed 2022-08-15 by BloodLemonS

darkness-gi, 2025-03-04:
> 已经解决了，感谢

你好，我有一个问题，就是作者说的这个路径Dynamic-VINS/blob/main/config/openloris/openloris_vio.yaml，我在Dynamic-VINS中没有找到blob这个文件夹，更别提后面的配置文件了，希望你能解答一下我的这个疑惑，感激不尽