HITSZ-NRSL/Dynamic-VINS #3: Nothing appears in rviz

QQQ300kuai, 2022-09-21:
[image]
line 3 `rosbag play YOUR_PATH_TO_DATASET/cafe.bag ` ，
i would like to know the path of cafe.bag,is cafe1-1.bag?or cafe1-2.bag?
thanks

jianhengLiu, 2022-09-21:
It's cafe1-1.bag or cafe1-2.
Sorry for unclear instruction.

QQQ300kuai, 2022-09-21:
[image]

Excuse me.
My problem is that I follow the instruction documentation but the trajectory is not displayed in rviz and there is no segmentation map.
`python YOUR_PATH_TO_DYNAMIC_VINS/scripts/merge_imu_topics.py cafe1-1.bag cafe1-2.bag`
`roslaunch vins_estimator openloris_vio_pytorch.launch`
`roslaunch vins_estimator vins_rviz.launch # Visualization`
`rosbag play YOUR_PATH_TO_DATASET/cafe.bag `

jianhengLiu, 2022-09-21:
I commented the yolo_ros relevant launch (line 5-7) in the openloris_vio_pytorch.launch.
Cancel the comment should be work.

jianhengLiu, 2022-09-21:
> It's cafe1-1.bag or cafe1-2. Sorry for unclear instruction.

apology,  should be cafe1-1-imu.bag with merge imu topics

QQQ300kuai, 2022-09-21:
I think there is a problem with the dataset in 3.1, I re-merge the gyro and accel topics into one.When re-executing the following command, it has been running for more than 10 minutes and has not stopped.Is this normal？
  `python YOUR_PATH_TO_DYNAMIC_VINS/scripts/merge_imu_topics.py cafe1-1.bag cafe1-2.bag`
[image]

jianhengLiu, 2022-09-21:
it seems normal.
perhaps printing out messages makes it slow, might try comment relevant 'print' code in the python file

closed 2022-09-26 by QQQ300kuai