HITSZ-NRSL/Dynamic-VINS #4: Custom bag file

eliabntt, 2022-10-07:
Hi there, we're running a custom bag file. 
My guess is that we need to change the focal length on the parameters.h file. 
Still we are not able to run with the IMU since we continuosly get ROS_INFO("Not enough features or parallax; Move device around");

Any tip on how we can solve it, apart for considering another bag? Any parameter we should look into given that we run at 240 Hz the IMU and have a different camera?

Thanks for your help

jianhengLiu, 2022-10-08:
The focal length problem could be refered to https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/issues/76.

Give a stable movement to the device or enable static_init in config file could solve the ROS_INFO problem.
https://github.com/HITSZ-NRSL/Dynamic-VINS/blob/524534f64e7577c77f5a8b0117a65918f94a30ee/config/realsense/vio_campus.yaml#L10

Modify parameters in config file according your own devices should work well.

Feel free to contact me, if there are any problem!

eliabntt, 2022-10-10:
Hi there, I reviewed the code a bit and saw that there are a numbers of FOCAL_LENGTHS used around and 460 (which is equal to the focal length value [here](https://github.com/HITSZ-NRSL/Dynamic-VINS/blob/6e6105da7b3b8caf53458d41743cd4215beb5991/vins_estimator/src/initial/solve_5pts.cpp#L216) and few lines below that, [here](https://github.com/HITSZ-NRSL/Dynamic-VINS/blob/6e6105da7b3b8caf53458d41743cd4215beb5991/vins_estimator/src/initial/initial_sfm.cpp#L159) but also in keyframe.cpp, feature_manager.cpp, and estimator.cpp). It's not clear to me if I need to change the one in parameters.h (which is const [here](https://github.com/HITSZ-NRSL/Dynamic-VINS/blob/6e6105da7b3b8caf53458d41743cd4215beb5991/vins_estimator/src/utility/parameters.h#L11)), and the one in keyframe.cpp [here](https://github.com/HITSZ-NRSL/Dynamic-VINS/blob/6e6105da7b3b8caf53458d41743cd4215beb5991/pose_graph/src/keyframe/keyframe.cpp#L181) or any of those 460.

I suspect that also has to do with the missing initialization.

jianhengLiu, 2022-10-10:
As mentioned [here](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/issues/76#issuecomment-544999597), 460 a virtual parameter and no need to change it.

You could turn on param `static_init` and `show_track` in the config file, check if the features in visualization are normal.
If  `static_init`  is false, it requires a movement of the device to initialize.
I also diable the code that deals with bleeding point few days ago(aaf7bde53712b6fc5459a5a24ddacc5e5d0fb81f), which might lead to dropping number of features. 
You might try pull the newest code, check if it work.
Sincerely sorry for that.

If it still not work, some visualization might help me to diagnose the problem. (try turn `show_track` on and have a screenshot on rviz) 
Thanks a lot.

eliabntt, 2022-10-11:
That parameter is a const double (NOT an extern one) however, so it stays 460 throughout all the runs. Moreover, i cannot find any place in which you overwrite it. 

Also, I'm more concerned about those 460s around the code.

Also, [this](https://github.com/HITSZ-NRSL/Dynamic-VINS/blob/6e6105da7b3b8caf53458d41743cd4215beb5991/pose_graph/src/keyframe/keyframe.cpp#L181) is definetly not virtual.

jianhengLiu, 2022-10-11:
I did not overwrite it indeed.
In my view, 460 is kind of hyper parameter for ransac threshold or parallax calculation.
Therefore, it might affect some of the features' tracking but a threshold not theirs projection location.
For example, the function `rejectWithF` firstly uses `liftProjective` to lift feature point from the image plane to the projective space with actual intrinsic, and then uses `FOCAL_LENGTH` project to a virtual camera (only focal length change) for subsequent RANSAC. 
It might help to adapt the system for more devices.
And I do think that replacing 460 with actual camera focal length would be beneficial.
While it may take some engineering work.

The code you mentioned is not used and looks like a redundant testing code.

Hope for more discussion, thank you!

eliabntt, 2022-10-31:
Seems working after these changes! Thanks!

closed 2022-10-31 by eliabntt