HITSZ-NRSL/Dynamic-VINS #5: Run error

QQQ300kuai, 2022-10-13:
When I run `roslaunch vins_estimator openloris_vio_pytorch.launch`, I encounter the following error：
[image]
Can you tell me how I should handle this.

jianhengLiu, 2022-10-14:
Check if you mistakenly change `#!/usr/bin/env python3` in `youlo_ros/src/demo_service_server.py`

QQQ300kuai, 2022-10-14:
I checked. Nothing changed.

jianhengLiu, 2022-10-14:
It seems like a python path problem.
You could try using conda to create a new environment to avoid this problem?

QQQ300kuai, 2022-10-14:
Again after installation, another error: `Cannot locate node of type yolo_trt_ros.py`
[image]

jianhengLiu, 2022-10-14:
It seems that you run [openloris_vio_tensorrt.launch](https://github.com/HITSZ-NRSL/Dynamic-VINS/blob/main/vins_estimator/launch/openloris/openloris_vio_tensorrt.launch) which requires tensorrt version of yolo_ros (`tensorrt branch`).
Please run [openloris_vio_pytorch.launch](https://github.com/HITSZ-NRSL/Dynamic-VINS/blob/main/vins_estimator/launch/openloris/openloris_vio_pytorch.launch).

QQQ300kuai, 2022-10-14:
I use conda to create a new environment,but the problem still exists.
> Again after installation, another error: `Cannot locate node of type yolo_trt_ros.py` [image]

QQQ300kuai, 2022-10-14:
I found the problem.
`#!/usr/bin/env python`   In ubuntu it becomes `#!/usr/bin/env python\r`
The solution:
`Vi` the file
Input `:set ff = unix`
Enter `:wq` to save the changes
Thank you!

closed 2022-10-14 by QQQ300kuai