HITSZ-NRSL/Dynamic-VINS #7: Some questions about the experiment

QQQ300kuai, 2022-10-24:
Excuse me.My question is as follows：

1.I run the office1-1 experiment, and use the evo tool with the` -r full` parameterthe to get `ATE RMSE` was 2.3, which was quite different from the results of the paper. I want to know whether ATE RMSE in Figure 6 is ATE obtained by considering both rotation and translation errors.In the evo tool, use the `-r full `parameter to test. Is that righ.And how can I improve to make the results close.

2.As shown in figure 6 each value in the upper left corner, is there a tool to test every scene average percentage of correct values？
Thank you!

jianhengLiu, 2022-10-24:
1. That's true, the evaluation metrics in paper are same as TUM RGB-D Datasets that only considers the translation errors.

2. To reproduce the kind of figures in paper, you could use the evaluation tools available at https://github.com/lifelong-robotic-vision/openloris-scene-tools

QQQ300kuai, 2022-10-24:
Sorry, I'm new to this tooI，and still have some questions.I used openloris-scene-tools, but some of the results were not available.Can you tell me if the following is true？And give me some advice.
1.  Take the following red box as an example.
   **95.7%** is the result obtained using openloris-scene-tools. **0.110, 0.040, 2.581** respectively represent ATE RMSE only considers the translation errors, only consider the RPE translation, only consider the RPE rotation.
2.  How can **0.040 and 2.581** be added to the graph since they cannot be obtained directly from openloris-scene-tools.
3.  I use openloris-scene-tools for operation, and the ATE RMSE of corridor and market sequence are 24.406 and 25.832, which are quite different from 2.367 and 1.185 in the paper. Can you tell me where I might be wrong？
[image]
Thank you~

jianhengLiu, 2022-10-25:
The t-rpe and r-rpe could be obtained from evaluate_rpe.py. 

The original [openloris_test_ros.py](https://github.com/lifelong-robotic-vision/openloris-scene-tools/blob/master/benchmark/openloris_test_ros.py) not provide per-seq testing tool and you might conduct a life-long testing. For more details, you could refer to paper 'Are We Ready for Service Robots? The OpenLORIS-Scene Datasets for Lifelong SLAM'.

And you could use the following code to conduct per-seq test.
```python
#!/usr/bin/env python3

# Copyright (C) <2019-2021> Intel Corporation
# SPDX-License-Identifier: MIT
# Authors: Siyuan Lu; Xuesong Shi

from __future__ import print_function

import argparse
from collections import OrderedDict
import glob
import logging
import psutil
#import rosbag
import rospy
import sys
import time
import threading
import subprocess
import yaml
import signal
import os

from geometry_msgs.msg import PoseStamped
from std_msgs.msg import UInt8
from std_msgs.msg import String

logger = logging.getLogger(__name__)
logger.setLevel(logging.INFO)
#formatter = logging.Formatter('%(asctime)s - %(levelname)s - %(message)s')
formatter = logging.Formatter('%(message)s')
ch = logging.StreamHandler()
ch.setFormatter(formatter)
logger.addHandler(ch)

init_pose_topic = '/openloris/reset_pose'
valid_frames = ['d400_color', 'd400_depth', 'd400_imu',
                't265_fisheye1', 't265_fisheye2', 't265_imu', 'base_link']

global pose_count
global current_pose

def pose_callback(data):
    # logger.info(data)
    print("%d.%d %s %f %f %f %f %f %f %f"
          % (data.header.stamp.secs, data.header.stamp.nsecs,
             rospy.get_time(),
             data.pose.position.x,
             data.pose.position.y,
             data.pose.position.z,
             data.pose.orientation.x,
             data.pose.orientation.y,
             data.pose.orientation.z,
             data.pose.orientation.w)
          )
    global pose_count, current_pose
    pose_count += 1
    current_pose = (data.pose.position.x,
                    data.pose.position.y,
                    data.pose.position.z,
                    data.pose.orientation.x,
                    data.pose.orientation.y,
                    data.pose.orientation.z,
                    data.pose.orientation.w)
[... 472 lines cut ...]
            2: (0.142683, 0.504664, 0.000000, 0.000000, 0.000000, 0.167331, 0.985901),
        },
        "corridor": {
            1: (0.000000, 0.000000, 0.000000, 0.000000, 0.000000, 0.000000, 1.000000),
            2: (9.014827, 36.233859, -0.000016, 0.000000, -0.000001, 0.084707, 0.996406),
            3: (9.296355, 36.612415, -0.000015, 0.000000, 0.000000, 0.085947, 0.996300),
            4: (21.175313, 47.339619, -0.000017, -0.000000, -0.000001, -0.762158, 0.647391),
            5: (6.816094, 36.074646, -0.000017, 0.000000, 0.000000, 0.117307, 0.993096),
        },
        "office": {
            1: (0.000000, 0.000000, 0.000000, 0.000000, 0.000000, 0.000000, 1.000000),
            2: (0.309635, 3.500538, 0.058010, 0.000811, 0.007326, -0.120414, 0.992696),
            3: (0.145258, 3.621898, 0.064342, 0.010893, -0.017257, 0.999785, -0.003740),
            4: (0.381370, 3.543278, 0.057061, 0.009458, -0.017024, 0.989809, -0.141061),
            5: (0.853251, 3.554253, 0.049278, 0.000541, -0.003027, -0.033718, 0.999427),
            6: (-0.105108, -0.001915, 0.001637, -0.000429, 0.004917, -0.024837, 0.999679),
            7: (0.690678, 3.603368, 0.055720, 0.001800, 0.011347, -0.198638, 0.980006),
        },
    },
}

if __name__ == "__main__":
    main()
    sys.exit(0)
```

closed 2022-11-04 by QQQ300kuai