HITSZ-NRSL/Dynamic-VINS #9: Initial orientation of the robot

eliabntt, 2022-10-31:
Hi there.
I'm facing some issue replaying my bags.
Essentially after some time the system starts drifting when I use VIO.
With VO it does not happen.
My guess is that it's mainly related to the fact that the robot is not horizontal and static at the beginning of the experiment.
Is there a way to say "this is my initial position/orientation" to your system such that the gravity and the Imu locations are initialized correctly?

eliabntt, 2022-10-31:
I've seen that you initialize it but it takes the average. Unfortunately my bag starts with the robot rotating in roll/pitch so I guess that that's the reason I see the drift.

Or maybe you've some other suggestion on why this might happen.

jianhengLiu, 2022-11-01:
If the initial state is not static, it would help to set `static_init` to 0, then it would follow the initialization procedure according to 'Online Temporal Calibration for Monocular Visual-Inertial Systems'.
The averages are mainly used to check if the imu is motivated or not and if not it is used to estimate the bias of accelerometor.
Another factor might come from the failure tracking of feature if the rotation is fierce or if it is pure rotation.
You could turn `show_track` on and check it in rviz.
So far we haven't have enough test on dynamic initialization, your feedback is beneficial to us.
Thanks a lot!

eliabntt, 2022-11-01:
Ok, I'll try to share something today maybe you'll have some ideas.
I've seen many people in the various vins version with problems similar to
mine.

Can you explain a bit more how static_init works? It was my understanding
that it was the transform between the Imu and the camera not between the
system and the map frame.

jianhengLiu, 2022-11-01:
The static_init is based on the prior that the `scale` is obtained from depth sensor or stereo, it give a prior to gravity direction that is horizontal and calculate the bias of imu.
They are both rough guess and will be refined in optimization.
We also tried to use the average to initialize, and didn't see a much different in our experiments. (`staticInitialAlignWithDepth` function in `estimator.cpp`)

closed 2022-11-01 by eliabntt