HITSZ-NRSL/Dynamic-VINS #11: [bug] exception row->0 columns->

eliabntt, 2022-11-14:
hi there

so sometimes (in VO mode) [this](https://github.com/HITSZ-NRSL/Dynamic-VINS/blob/675722603cace8a4c118e79540d9354694b058f9/vins_estimator/src/factor/marginalization_factor.cpp#L276) lines throws an exception.

I've checked VINS FUSION and they check if m == 0 and set a param (`valid`) to false. That is then used in the `estimator.cpp` file ([here](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/blob/master/vins_estimator/src/estimator/estimator.cpp#L1045) [here](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/blob/master/vins_estimator/src/estimator/estimator.cpp#L1145) and [here](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/blob/master/vins_estimator/src/estimator/estimator.cpp#L1265)) 

Did this ever happen to you?

jianhengLiu, 2022-12-03:
Sorry for the late reply. I've been caught up in other things.
The vo mode didn't have a throughout examination; it may have bugs; sorry for that.
I will check it later when I am available. Or you could issue a pull request if you fix it.
 Thanks a lot!

eliabntt, 2022-12-03:
We simply added vins fusion code. This happens when there's no features to track.

closed 2022-12-03 by eliabntt