HITSZ-NRSL/Dynamic-VINS #13: Yolo v3 or v5?

eliabntt, 2022-12-16:
Hi,
in the paper you mention it is yolov3, then in the credits of the readme you say yolov5 (and yolo_ros uses that afaik), but the weights are from v3. Which one is it? I am a bit confused.

jianhengLiu, 2022-12-17:
In the paper, the implements are based on yolov3 that run on embedding platforms (NVIDIA Xavier(tensorrt ver.) and HUAWEI Atlas(ascend ver.)).
Beyond the paper, we modified the popular and developed yolov5 in a normal PC platform.
Normally, yolov3 and yolov5 shall make no big difference in the results of Dynamic-VINS.

eliabntt, 2022-12-17:
Modified as in? Any other difference with the paper? I ask because we want
to use you in our work so we need to be precise on what we do

jianhengLiu, 2022-12-17:
Modification just for ros communication.
No other difference with the paper

 
 
 &gt; In the paper, the implements are based on yolov3 that run on embedding
 &gt; platforms (NVIDIA Xavier(tensorrt ver.) and HUAWEI Atlas(ascend ver.)).
 &gt; Beyond the paper, we modified the popular and developed yolov5 in a normal
 &gt; PC platform.
 &gt; Normally, yolov3 and yolov5 shall make no big difference in the results of
 &gt; Dynamic-VINS.
 &gt;
 &gt; —
 &gt; <https://github.com/HITSZ-NRSL/Dynamic-VINS/issues/13#issuecomment-1355968198&gt;,
 &gt; .
 &gt; ***@***.***&gt;
 &gt;

closed 2023-02-28 by eliabntt