HITSZ-NRSL/Dynamic-VINS #15: The main function of source code and the figure of OpenLORIS

HJMGARMIN, 2023-03-03:
Great work!
I met some problems when I read the paper and run it.
1. How do you draw the below picture of OpenLORIS? We also want to draw a similar figure to evaluate the robustness of our algorithm.
[image]

2. Can you tell me where is the main function of the source code? I can not find the main() of estimator. Thank you very much.

jianhengLiu, 2023-03-03:
1. To draw the figure, you could ref to the [openloris-scene-tools](https://github.com/lifelong-robotic-vision/openloris-scene-tools/tree/master/benchmark)
2. It's a multi-thread software, the main() is seperated to 2 main thread process. For tracker ref to [estimator_nodelet.cpp#L245](https://github.com/HITSZ-NRSL/Dynamic-VINS/blob/main/vins_estimator/src/estimator_nodelet.cpp#L245), for estimator ref to [estimator_nodelet.cpp#L245](https://github.com/HITSZ-NRSL/Dynamic-VINS/blob/main/vins_estimator/src/estimator_nodelet.cpp#L245)

HJMGARMIN, 2023-03-03:
Thank you for your response.
I found that the process_tracker thread was started in "trackThread   = std::thread(&EstimatorNodelet::process_tracker, this);" of "void onInit() override". However, how the "onInit()" function works in the project? I do not know where is it called.

jianhengLiu, 2023-03-03:
just ignore "onInit() override"

 

 I found that the process_tracker thread was started in "trackThread   = std::thread(&amp;EstimatorNodelet::process_tracker, this);" of "void onInit() override". However, how the "onInit()" function works in the project? I do not know where is it called.

closed 2023-03-03 by HJMGARMIN