HITSZ-NRSL/Dynamic-VINS #16: The format of trajectory saved

HJMGARMIN, 2023-03-04:
Hello, I have run Dynamic-VINS on my platform successfully. However, I met a problem when I evaluate it.
I do not know how to save the estimation trajectory as the format of TUM. The saved *.csv file's format is different from the grountruth.txt of OpenLORIS.

HJMGARMIN, 2023-03-06:
Hello, can you help me?

jianhengLiu, 2023-03-06:
You could use evaluation tools: [evo](https://github.com/MichaelGrupp/evo) or [openloris-scene-tools](https://github.com/lifelong-robotic-vision/openloris-scene-tools)

HJMGARMIN, 2023-03-06:
I use evo to evaluate it, but I got below issuses:

evo_ape tum /media/ubuntu-slam/10B45E5AB45E4280/dataset/OpenLORIS/cafe/cafe1-1/groundtruth.txt vins_result_loop.csv -va
--------------------------------------------------------------------------------
Loaded 2281 stamps and poses from: /media/ubuntu-slam/10B45E5AB45E4280/dataset/OpenLORIS/cafe/cafe1-1/groundtruth.txt
[ERROR] TUM trajectory files must have 8 entries per row and no trailing delimiter at the end of the rows (space)

jianhengLiu, 2023-03-06:
As I remember, openloris saved format adds another entry, therefore, could not directly use evo to evaluate openloris' saving poses.
More details, please ref to the openloris-tools codes.

HJMGARMIN, 2023-03-06:
Thanks for your reponse. But the groundtruth.txt of OpenLORIS has the same format as TUM RGB-D. I can evaluate ORB-SLAM3 on OpenLORIS using evo.
Additionally, for TUM RGB-D, I found the saved files from Dynamic-VINS adds another entry. There are data except time,x,y,z,qw,qx,qy,qz.
How do you evaluate Dynamic-VINS on TUM RGB-D dataset?

HJMGARMIN, 2023-03-06:
I have solved above problems.
Now, I meet a new problem. The saved file often contains nothing after running Dynamic-VINS. I need to run a sequence two or three times to get a *.csv that contains estimation poses.
[image]

closed 2023-03-06 by HJMGARMIN

zhixuan4123, 2023-09-24:
> I have solved above problems. Now, I meet a new problem. The saved file often contains nothing after running Dynamic-VINS. I need to run a sequence two or three times to get a *.csv that contains estimation poses. [image]

Has this problem been solved yet? What was the cause?

yg-0527, 2023-11-17:
> > 我已经解决了上述问题。现在，我遇到了一个新问题。运行 Dynamic-VINS 后，保存的文件通常不包含任何内容。我需要运行序列两到三次才能获得包含估计姿势的 *.csv。[image]
> 
> 这个问题解决了吗？原因是什么？

请问你解决这个轨迹输出的问题了吗？我也碰到这个问题了，希望能沟通交流一下！

yg-0527, 2023-11-18:
> Thanks for your reponse. But the groundtruth.txt of OpenLORIS has the same format as TUM RGB-D. I can evaluate ORB-SLAM3 on OpenLORIS using evo. Additionally, for TUM RGB-D, I found the saved files [...]

Hello, may I ask how you resolved the issue of '[ERROR] TUM trajectory files must have 8 entries per row and no tracking delimiter at the end of the rows (space)'? Can you please let me know? I also encountered the same problem.