HITSZ-NRSL/Dynamic-VINS #17: undefined symbol: _ZN6Sophus3SO3C1Eddd

LYF-hahaha, 2023-03-07:
作者你好！

我在尝试按readme中的内容复现你的工作，目前遇到一个问题
在按如下指令运行cafe1-1数据集的时候出现报错undefined symbol: _ZN6Sophus3SO3C1Eddd（如下面截图）
看内容感觉与sophus有关，但sophus我也是按照git checkout a621ff 的要求安装的

希望能获得指点，感谢~

`roslaunch vins_estimator openloris_vio_pytorch.launch`
`roslaunch vins_estimator vins_rviz.launch`
`rosbag play ~/Dynamic_VINS/Dataset/OpenLORIS/cafe/cafe1-1.bag-imu.bag`

[image]

HJMGARMIN, 2023-03-08:
把OpenCV版本换成ROS自带的试试

FanJinlong2023, 2024-04-01:
> 把OpenCV版本换成ROS自带的试试

换成ros的opencv后，还是出现错误

FanJinlong2023, 2024-04-01:
[image]

closed 2024-05-22 by jianhengLiu