HITSZ-NRSL/Dynamic-VINS #20: Failed to load nodelet [/EstimatorNodelet] of type [vins_estimator/EstimatorNodelet] even after refreshing the cache

YongshuaiZ, 2023-03-25:
作者您好，当我运行时，出现如下问题：
提示我缺少libestimator_nodelet.so文件，但是我的/devel/lib/目录下已有该文件。我的运行环境为ubuntu20.04 ros noetic 

[image]
[ INFO] [1679707768.981822341]: waitForService: Service [/yolo_service] has not been advertised, waiting...
[ INFO] [1679707768.984874317]: Loading nodelet /EstimatorNodelet of type vins_estimator/EstimatorNodelet to manager nodelet_manager_pc with the following remappings:
[ INFO] [1679707768.985121969]: /camera/color/image_raw -> /d400/color/image_raw
[ INFO] [1679707768.985669141]: waitForService: Service [/nodelet_manager_pc/load_nodelet] has not been advertised, waiting...
[ INFO] [1679707768.989190628]: Initializing nodelet with 20 worker threads.
[ INFO] [1679707769.006713921]: waitForService: Service [/nodelet_manager_pc/load_nodelet] is now available.
[ERROR] [1679707769.030456673]: Failed to load nodelet [/EstimatorNodelet] of type [vins_estimator/EstimatorNodelet] even after refreshing the cache: Failed to load library /home/zhu/SlamProjects/Dynamic_VINS_ws/devel/lib//libestimator_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /home/zhu/SlamProjects/Dynamic_VINS_ws/devel/lib//libestimator_nodelet.so: undefined symbol: _Z11DEPTH_TOPICB5cxx11)
[ERROR] [1679707769.036358328]: The error before refreshing the cache was: Failed to load library /home/zhu/SlamProjects/Dynamic_VINS_ws/devel/lib//libestimator_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /home/zhu/SlamProjects/Dynamic_VINS_ws/devel/lib//libestimator_nodelet.so: undefined symbol: _Z11DEPTH_TOPICB5cxx11)
[FATAL] [1679707769.056044454]: Failed to load nodelet '/EstimatorNodelet` of type `vins_estimator/EstimatorNodelet` to manager `nodelet_manager_pc'
[EstimatorNodelet-5] process has died [pid 19624, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load vins_estimator/EstimatorNodelet nodelet_manager_pc /camera/color/image_raw:=/d400/color/image_raw __name:=EstimatorNodelet __log:=/home/zhu/.ros/log/7c3e0fb6-caac-11ed-b95c-c1c703164ee1/EstimatorNodelet-5.log].
log file: /home/zhu/.ros/log/7c3e0fb6-caac-11ed-b95c-c1c703164ee1/EstimatorNodelet-5*.log
YOLOv5 🚀 e310fa9 torch 1.10.2+cu113 CUDA:0 (NVIDIA GeForce RTX 3060, 12036MiB)

jianhengLiu, 2023-03-25:
参考 #1

YongshuaiZ, 2023-03-25:
感谢回复，我指定了cv_bridge中opencv的版本以及项目中的opencv版本后，解决了问题。

closed 2023-03-25 by YongshuaiZ