HITSZ-NRSL/Dynamic-VINS #22: 运行HITSZ & THUSZ Datasets出错

theyunfeng, 2023-04-19:
在跑hitsz_00.bag时，在运行roslaunch vins_estimator realsense_vio_campus.launch 的终端里出现
[image]，
在rviz中有warning
[image]
请问这是什么原因呢？

jianhengLiu, 2023-04-20:
可能是opencv版本冲突？
默认是使用ubuntu 20.04，ROS noetic自带的opencv 4.2.0以及python3
确保编译和运行使用的是同一个python

theyunfeng, 2023-04-20:
> 可能是opencv版本冲突？ 默认是使用ubuntu 20.04，ROS noetic自带的opencv 4.2.0以及python3 确保编译和运行使用的是同一个python

会是单独运行hitsz_00.bag的原因吗？

jianhengLiu, 2023-04-20:
> > 可能是opencv版本冲突？ 默认是使用ubuntu 20.04，ROS noetic自带的opencv 4.2.0以及python3 确保编译和运行使用的是同一个python
> 
> 会是单独运行hitsz_00.bag的原因吗？

应该不是，只是把一个大数据分段了，数据应该没有影响

theyunfeng, 2023-04-20:
ok,多谢，看样子是链接库版本不对，让cv_bridge强制链接自己的opencv就好了，虽然还是有warning

zhixuan4123, 2023-07-19:
> 请问这个Upsample是怎么解决的尼？

closed 2024-11-12 by jianhengLiu