HITSZ-NRSL/Dynamic-VINS #23: 运行campus数据集时没有显示feature_img

V1ct0ryan, 2023-04-19:
[image]
我把yolo需要的环境也配置了，但是只能追踪到imu数据没有相机的目标检测和特征点检测，请问一下作者如何得到如你们所传视频的可视化效果呢，谢谢

theyunfeng, 2023-04-19:
> [image] 我把yolo需要的环境也配置了，但是只能追踪到imu数据没有相机的目标检测和特征点检测，请问一下作者如何得到如你们所传视频的可视化效果呢，谢谢

我的更离谱，一开始报opencv的错误，重新catkin_make后不报错了但是rviz里除了fixed_frame和tracked_imgae那里一直是黄色的警告，说是没有受到信息，哥们，能问一下你的ros版本、opencv版本吗？

jianhengLiu, 2023-04-20:
> [image] 我把yolo需要的环境也配置了，但是只能追踪到imu数据没有相机的目标检测和特征点检测，请问一下作者如何得到如你们所传视频的可视化效果呢，谢谢

配置文件设置`show_track: 1`，图片发布在`/vins_estimator/semantic_mask`节点上

V1ct0ryan, 2023-04-21:
> > [image] 我把yolo需要的环境也配置了，但是只能追踪到imu数据没有相机的目标检测和特征点检测，请问一下作者如何得到如你们所传视频的可视化效果呢，谢谢
> 
> 我的更离谱，一开始报opencv的错误，重新catkin_make后不报错了但是rviz里除了fixed_frame和tracked_imgae那里一直是黄色的警告，说是没有受到信息，哥们，能问一下你的ros版本、opencv版本吗？

我的是noetic版本，opencv就是按照requirements里安装的

V1ct0ryan, 2023-04-21:
> > [image] 我把yolo需要的环境也配置了，但是只能追踪到imu数据没有相机的目标检测和特征点检测，请问一下作者如何得到如你们所传视频的可视化效果呢，谢谢
> 
> 配置文件设置`show_track: 1`，图片发布在`/vins_estimator/semantic_mask`节点上

十分感谢，问题解决了

closed 2023-05-25 by V1ct0ryan