HITSZ-NRSL/Dynamic-VINS #25: 运行过程出错：unstable features tracking, please slowly move you device!

bzwwww, 2023-04-27:
按github中配置运行openloris和thusz数据集时都会报标题中的错误。
[image]
[image]

jianhengLiu, 2023-04-27:
可能是后端特征点估计飘了，可以分别尝试
1. 不使用静止初始化`static_init: 0`
2. 调大深度置信区间`depth_max_dist: 6`
3. 增大特征点数量`max_cnt: 130`

bzwwww, 2023-04-27:
我采用的是config中原始配置，并且尝试将 depth_max_dist: 6 更改  都没有成功

jianhengLiu, 2023-04-28:
我这边用默认配置测试了一下好像都还正常
openloris: `roslaunch vins_estimator openloris_vio_pytorch.launch`
thusz: `roslaunch vins_estimator realsense_vio_campus.launch`

如果已经pull了最新版还是有问题的话，不清楚是不是这个commit(4e73b9e0806bbd094e7bedfadf4d5ebb402a27e4)可能版本有点差异？可以尝试改回来试试

bzwwww, 2023-04-28:
我之前运行的版本中是cv::SOLVEPNP_ITERATIVE，我重新clone了一下最新的版本，重新运行，还是会发现上述错误，但出现了不同的情况。
track_img中不再是深度图，同时我使用的不是ros-noetic中自带的OpenCV，我使用的版本是OpenCV4.5.5。您认为这两者中有关系吗？感谢您的帮助！
[image]
[image]

jianhengLiu, 2023-04-28:
有可能，我使用的是ros自带的opencv4.2.0
rviz显示不同可能是更改了topic，在display里重新选下应该就好

bzwwww, 2023-04-28:
我重新编译了OpenCV4.2.0的版本，已经可以成功运行，感谢您的帮助！

yg-0527, 2023-06-05:
你好，请问这个一直下载不了源码是怎么回事啊？我网络良好，也试过开vpn，都显示网页404。请问谁可以分享一下源码吗？万分感谢！！！！

bzwwww, 2023-08-23:
你好，还想问一下这份代码里还有闭环检测的功能吗，我拿这套代码跑kitti数据集时出现了比较大的偏差，想尝试开闭环之后试试

jianhengLiu, 2024-11-12:
应该是launch文件中注释了回环的节点，有需要的话，取消注释应该就能用

closed 2024-11-12 by jianhengLiu