HITSZ-NRSL/Dynamic-VINS #26: ubuntu18.04环境配置

fishZeng, 2023-05-17:
我这边是ros melodic+opencv3.2+cuda11.0，反复尝试之后还是看不到目标检测的结果。请问当时在ubuntu18.04下运行时使用的各种包的版本是多少？

wsanm, 2023-05-30:
> 不好意思，后面开发大部分都是在20.04上完成的，可以自行参照下提示/报错等排查一下，麻烦了 我估计应该是使用的opencv4.2

您好，我18.04运行成功。问下之前的issue不是用自己安装的opencv会报错吗？

closed 2023-06-01 by fishZeng

yg-0527, 2023-08-02:
> 现报错，我尝试下换opencv4.2看看
你好，想问下你在ubuntu18上配置成功了吗？我这边配置到最后一步的时候出错了，输入指令：pip3 install -r yolo_ros/requirements.txt   最后报错：
[image]

fishZeng, 2023-08-04:
> > 现报错，我尝试下换opencv4.2看看
> > 你好，想问下你在ubuntu18上配置成功了吗？我这边配置到最后一步的时候出错了，输入指令：pip3 install -r yolo_ros/requirements.txt   最后报错：
> > [image]

你试试直接下载opencv-python到本地再安装，我这一步没出现问题。最后用tensorrt版本的yolo配置成功了，用自带的opencv3.2版本，你可以参考tensorrt的版本配置一下试试

yg-0527, 2023-08-05:
感谢您的回复，我重新安装配置了环境，没有出现之前的“failed to build opencv-python”问题，但是在运行的时出现了问题：
我打开三个终端，输入官网上再pytorch环境下运行的指令，我的指令分别是（第一张图）：
roslaunch vins_estimator openloris_vio_pytorch.launch
roslaunch vins_estimator vins_rviz.launch 
rosbag play /home/yugui/code/catkin_ws/datas/cafe1-1.bag

运行之后rviz那边什么都没显示，轨迹、坐标、跟踪图像都没显示 （第二张图）。

我不清楚是不是我没切换坐标，所以我将FixedFrame 从“map”  切换成 “laser” ，但是rviz还是没反应，只是在图二中的两处标红的地方变成了一处警告和正常（第三张图）。
我按照官网上的方式解压了cafe1-1_2.rosbag.tar 文件，如第四张图所示，最终得到（2个.bag文件，2个bag-imu.bag文件和2个orig.bag文件）。我运行的文件是cafe1-1.bag。
我不清楚是不是我运行的数据集有问题还是其他地方出现错误了。
您知道怎么解决吗？或者您碰到过类似的问题吗？再次感谢您的阅读和回复！！

Sl eeping
***@***.***

抄送: "Sl ***@***.******@***.***&gt;;

fishZeng, 2023-08-07:
> 感谢您的回复，我重新安装配置了环境，没有出现之前的“failed to build opencv-python”问题，但是在运行的时出现了问题： 我打开三个终端，输入官网上再pytorch环境下运行的指令，我的指令分别是（第一张图）： roslaunch vins_estimator openloris_vio_pytorch.launch roslaunch vins_estimator vins_rviz.launch  rosbag play /home/yugui/code/catkin_ws/datas/cafe1-1.bag 运行之后rviz那边什么都没显示，轨迹、坐标、跟踪图像都没显示 （第二张图）。 我不清楚是不是我没切换坐标，所以我将FixedFrame 从“map”  切换成 “laser” ，但是rviz还是没反应，只是在图二中的两处标红的地方变成了一处警告和正常（第三张图）。 我按照官网上的方式解压了cafe1-1_2.rosbag.tar 文件，如第四张图所示，最终得到（2个.bag文件，2个bag-imu.bag文件和2个orig.bag文件）。我运行的文件是cafe1-1.bag。 我不清楚是不是我运行的数据集有问题还是其他地方出现错误了。 您知道怎么解决吗？或者您碰到过类似的问题吗？再次感谢您的阅读和回复！！ Sl eeping ***@***.***  
> […](#)

应该运行bag-imu.bag，原始bag不能直接跑出结果

yg-0527, 2023-08-07:
谢谢你的回复，确实如你所说得运行bag-imu.bag文件。但是我在运行文件之后，rviz中的Odometry中出现Status：Error错误，请问你知道是什么原因？需要怎么去解决呢？

Sl eeping
***@***.***

fishZeng, 2023-08-08:
> 谢谢你的回复，确实如你所说得运行bag-imu.bag文件。但是我在运行文件之后，rviz中的Odometry中出现Status：Error错误，请问你知道是什么原因？需要怎么去解决呢？ Sl eeping ***@***.***  
> […](#)

搜索下吧，这个感觉是某些比较常见的问题，我没印象了

bochunyang, 2024-05-27:
> > 不好意思，后面开发大部分都是在20.04上完成的，可以自行参照下提示/报错等排查一下，麻烦了 我估计应该是使用的opencv4.2
> 
> 运行的时候没出现报错，我尝试下换opencv4.2看看

> > 不好意思，后面开发大部分都是在20.04上完成的，可以自行参照下提示/报错等排查一下，麻烦了 我估计应该是使用的opencv4.2
> 
> 运行的时候没出现报错，我尝试下换opencv4.2看看

> > 不好意思，后面开发大部分都是在20.04上完成的，可以自行参照下提示/报错等排查一下，麻烦了 我估计应该是使用的opencv4.2
> 
> 运行的时候没出现报错，我尝试下换opencv4.2看看

您好，我这边复现遇到了和您一样的问题，也是在Ubuntu18.04上可视化无法出现检测对象和semantic_mask，请问您当时替换成opencv4.2.0后有没有出现结果呢，我这边替换了opencv4.2.0，相应的cv_bridge也适配到，也没有出现nodelet报错，但还是没有出现检测对象框和semantic_mask。