HITSZ-NRSL/Dynamic-VINS #27: 在Jetson xavier nx上运行时报错：Failed to load nodelet '/EstimatorNodelet` of type `vins_estimator/EstimatorNodelet` to manager `nodelet_manager'

fishZeng, 2023-06-02:
提示信息如下：
`[TensorRT] WARNING: Using an engine plan file across different models of devices is not recommended and is likely to affect performance or even cause errors.`
`bingding: data (3, 640, 640)`
`bingding: prob (6001, 1, 1)`
`batch size is 1`
`[FATAL] [1685328734.855312649]: Failed to load nodelet '/EstimatorNodelet` of type `vins_estimator/EstimatorNodelet` to manager `nodelet_manager'`
`[nodelet_manager-3] process has died [pid 7843, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=nodelet_manager __log:=/home/wheeltec/.ros/log/b508d36a-fdcb-11ed-812a-5c879c1d9813/nodelet_manager-3.log].`
`log file: /home/wheeltec/.ros/log/b508d36a-fdcb-11ed-812a-5c879c1d9813/nodelet_manager-3*.log`
`[EstimatorNodelet-4] process has died [pid 7844, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load vins_estimator/EstimatorNodelet nodelet_manager __name:=EstimatorNodelet __log:=/home/wheeltec/.ros/log/b508d36a-fdcb-11ed-812a-5c879c1d9813/EstimatorNodelet-4.log].`
`log file: /home/wheeltec/.ros/log/b508d36a-fdcb-11ed-812a-5c879c1d9813/EstimatorNodelet-4*.log
`

kukulinxin, 2023-11-28:
想问一下你是怎么解决的？

fishZeng, 2023-11-28:
> 想问一下你是怎么解决的？

我也忘了，最后直接用电脑当上位机跑的