HITSZ-NRSL/Dynamic-VINS #37: Can you provide me with detailed materials about the added IMU-aided feature tracking?

Eunseong17, 2023-11-08:
Thank you for your great work. 
I am wondering about IMU-aided feature tracking. I searched for this reference in your paper (Dynamic-VINS), but I couldn't find it.

jianhengLiu, 2023-11-08:
[image]
[image]

Eunseong17, 2023-11-08:
Thank you for the fast response, but these equations are a little hard to understand. Could you provide a brief explanation or a reference paper?

Eunseong17, 2023-11-15:
I have additional question.

1. In the equation, does 'sin(θ)*n' imply 'sin(θ)*n^T'?
2. Could you explain what 'z' represents in this context?
3. Also, it appears that the equations only account for rotation. Is translation (straight-line motion) not considered in this model?

Thank you.

closed 2024-03-15 by Eunseong17