HITSZ-NRSL/Dynamic-VINS #39: ground truth of HITSZ & THUSZ Datasets

Mahmoudadham4040, 2023-11-14:
could you please provide us with the ground truth or the RTK-GPS trajectories for the  HITSZ & THUSZ Datasets ?

jianhengLiu, 2023-11-14:
Sorry that these datasets do not contain GT or GPS trajectory.
It is collected for qualitative experiments, and you could compare results with loop-closure trajectory or satellite map.

Mahmoudadham4040, 2023-11-14:
Thanks for your consideration.
I have another point, I don't have depth image messages and want to evaluate Vins with yolo? could you please provide me your suggestions to edit in this code?

jianhengLiu, 2023-11-14:
One plausible way is to modify codes to add choice to receive datastream without depth, and mask all features in yolo bbox

closed 2024-03-31 by jianhengLiu