HITSZ-NRSL/Dynamic-VINS #41: vins_result_no_loop.scv无内容

yg-0527, 2023-11-16:
作者您好，我在跑tum-rgbd-fr3-walking-halfsphere数据集时，运行结束后，在output文件输出的vins_result_no_loop.scv文件是空白内容，请问您能指点一下我是哪里可能有问题吗？我已经修改了tum的yaml文件：
[image]
[image]
[image]
[image]

Suhao07, 2024-09-01:
问问这个问题解决了吗？

explorer-000, 2024-12-10:
请问这个问题解决了吗

explorer-000, 2024-12-10:
我修改yaml文件后重新编译就可以了

explorer-000, 2024-12-10:
但是我们raw_image显示不了，请问楼主使用的rosbag数据包吗

closed 2025-02-11 by jianhengLiu