HITSZ-NRSL/Dynamic-VINS #43: Failed to load nodelet '/EstimatorNodelet of type vins_estimator/EstimatorNodelet

kukulinxin, 2023-11-29:
你好，我同样遇到了Failed to load nodelet '/EstimatorNodelet of type vins_estimator/EstimatorNodelet to manager nodelet_manager_pc'的问题，但是由于我是在NVIDIA NX上部署，NX使用SDK manager装机时自带了opencv4，安装ros-melodic时自带的opencv并没有安装，所以无法向常规的那样指定ros自带的opencv。我想询问的是：导致节点无法加载的具体原因是什么？自己下载opencv 3.2.0进行编译（ros-melodic中自带的版本），再在cv_bridge中修改可以吗？谢谢。

jianhengLiu, 2024-05-22:
可以的
也可以修改cv_bridge的版本，重新安装适用opencv4的版本

Suhao07, 2024-08-31:
请问这个报错是怎么解决的？能详细说一下吗？

kukulinxin, 2024-09-01:
> 请问这个报错是怎么解决的？能详细说一下吗？

你好，后来是通过安装适合opencv4的cv_bridge包来解决的，机子不在身边，具体有些忘了，你可以先尝试一下。

Suhao07, 2024-09-01:
请问你是怎么运行得到semantic image的？我已经解决了那个opencv的问题，成功运行了代码。但是按照readme运行，rviz里没有tracked image （显示没有收到图片信息）。我运行的是openloris数据集，请问需要修改什么参数吗？发自我的手机-------- 原始邮件 --------发件人： kukulinxin ***@***.***>日期： 2024年9月1日周日 下午3:55收件人： HITSZ-NRSL/Dynamic-VINS ***@***.***>抄送： rickyman ***@***.***>, Comment ***@***.***>主    题： Re: [HITSZ-NRSL/Dynamic-VINS] Failed to load nodelet '/EstimatorNodelet of type vins_estimator/EstimatorNodelet (Issue #43)

请问这个报错是怎么解决的？能详细说一下吗？

你好，后来是通过安装适合opencv4的cv_bridge包来解决的，机子不在身边，具体有些忘了，你可以先尝试一下。

closed 2025-02-11 by jianhengLiu