HITSZ-NRSL/Dynamic-VINS #47: NVIDIA Jeston Xavier NX 上部署的一些问题

kukulinxin, 2023-12-19:
作者您好，我尝试了在NVIDIA Jeston Xavier NX上部署Dynamic-VINS，相机使用的是Intel RealSense D435i，但目前发现的是图像的帧率比较低，其次是只要IMU很容易漂移。针对图像的帧率比较低的问题，我想问的是作者您有使用GPU加速吗，还是说原文中无人机使用的NVIDIA Jetson AGX Xavier 的CPU能力足够强没有这个问题。针对漂移问题，我使用的是D435i自带的IMU，会不会是因为D435i的IMU精度不够，又或是其他什么原因？希望作者能给予一定建议和传授经验。

jianhengLiu, 2023-12-19:
1. 帧率低可以通过降低特征点数量来提高，目前定位方法只用了 cpu 。当然现在也有很多基于 gpu 的特征提取方法，应该能够大幅提速。
帧率也有可能受限于 yolo 部分，因为需要对检测结果进行同步，你可以看一下哪部分是瓶颈，来针对性做优化
2. Imu 精度不够可以尝试增大 imu 相关的不确定度项。这个也和使用场景有关，如果手持的话，震动没那么大一般好点

Sent at: 2023年12月19日(Tue) Noon1:23
To: ***@***.***&gt;;
Cc: ***@***.***&gt;;
Subject: [HITSZ-NRSL/Dynamic-VINS] NVIDIA Jeston Xavier NX 上部署的一些问题 (Issue #47)

kukulinxin, 2023-12-19:
> 1. 帧率低可以通过降低特征点数量来提高，目前定位方法只用了 cpu 。当然现在也有很多基于 gpu 的特征提取方法，应该能够大幅提速。 帧率也有可能受限于 yolo 部分，因为需要对检测结果进行同步，你可以看一下哪部分是瓶颈，来针对性做优化 2. Imu 精度不够可以尝试增大 imu 相关的不确定度项。这个也和使用场景有关，如果手持的话，震动没那么大一般好点
> […](#)

感谢您宝贵的回复和建议，我进一步调试优化试试。

kukulinxin, 2023-12-22:
> 1. 帧率低可以通过降低特征点数量来提高，目前定位方法只用了 cpu 。当然现在也有很多基于 gpu 的特征提取方法，应该能够大幅提速。 帧率也有可能受限于 yolo 部分，因为需要对检测结果进行同步，你可以看一下哪部分是瓶颈，来针对性做优化 2. Imu 精度不够可以尝试增大 imu 相关的不确定度项。这个也和使用场景有关，如果手持的话，震动没那么大一般好点
> […](#)

作者您好，我后来进一步在检测上进行调整，当yolo检测速度提升后，SLAM视觉里程计的tracking的帧率确实有明显提高。但是定位的漂移还是很大（手持），稍微晃一点位姿就开始乱飞了，我尝试在配置文件中把imu关了进行测试，虽然定位精度不一定准确但是不怎么存在漂移，所以感觉还是IMU的问题，目前分析有几种原因：1）D435i给的imu数据不够准确。2）imu标定有问题。因此想再问问作者，你们在真机测试的IMU数据是来自D435i还是无人机上的；如果是imu标定不够精准真的会造成这么大的漂移吗？或者其他的我不知道的原因，我在VINS使用方面可能经验还是太少了，感谢。

jianhengLiu, 2023-12-22:
有可能外参不对，你可以自行在 rviz 显示出各个坐标系，看一下对不对
真机使用的飞控 imu 数据
再进一步可以自行尝试调节一下 imu parameters，找到合适的参数

Sent at: 2023年12月22日(Fri) PM5:21

kukulinxin, 2023-12-22:
好的，感谢。再顺便疑问这会和D435i的imu输出频率有关吗？

kukulinxin, 2023-12-22:
> 有可能外参不对，你可以自行在 rviz 显示出各个坐标系，看一下对不对 真机使用的飞控 imu 数据 再进一步可以自行尝试调节一下 imu parameters，找到合适的参数
> […](#)

好的，感谢。再顺便疑问这会和D435i的imu输出频率有关吗？

kukulinxin, 2023-12-22:
> 有可能外参不对，你可以自行在 rviz 显示出各个坐标系，看一下对不对 真机使用的飞控 imu 数据 再进一步可以自行尝试调节一下 imu parameters，找到合适的参数
> […](#)

好的，感谢。

kukulinxin, 2023-12-22:
> 有可能外参不对，你可以自行在 rviz 显示出各个坐标系，看一下对不对 真机使用的飞控 imu 数据 再进一步可以自行尝试调节一下 imu parameters，找到合适的参数
> […](#)

想再问问作者您，是否有使用unsynchronization parameters，我记得D435i已经同步了，但是也有人说有误差。

closed 2025-02-11 by jianhengLiu