HITSZ-NRSL/Dynamic-VINS #51: 你好 我运行的时候有点问题

tulingcheng86, 2024-05-21:
我是按照https://github.com/HITSZ-NRSL/Dynamic-VINS/blob/main/doc/INSTALL.md#21-nvidia-devices构建
电脑运行；
yolo_ros单独运行没问题，`roslaunch yolo_ros yolo_service.launch`和`rosbag  play cafe1-1.bag-imu.bag`运行有检测结果；
但是运行
```
roslaunch vins_estimator openloris_vio_pytorch.launch
roslaunch vins_estimator vins_rviz.launch # Visualization
rosbag play  rosbag  play cafe1-1.bag-imu.bag
```
rviz里面没画面和没轨迹

[image]

在我运行`roslaunch vins_estimator openloris_vio_pytorch.launch`
```
... logging to /root/.ros/log/a90bea7e-1766-11ef-8972-0242ac110002/roslaunch-d22a80a4bf79-149641.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://d22a80a4bf79:39177/

SUMMARY
========

PARAMETERS
 * /EstimatorNodelet/config_file: /root/catkin_ws/s...
 * /EstimatorNodelet/vins_folder: /root/catkin_ws/s...
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /yolov5/action: False
 * /yolov5/device: gpu
 * /yolov5/img_size: 640

NODES
  /
    EstimatorNodelet (nodelet/nodelet)
    demo_server (yolo_ros/demo_service_server.py)
    nodelet_manager_pc (nodelet/nodelet)
    service_client (yolo_ros/service_client)

auto-starting new master
process[master]: started with pid [149654]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a90bea7e-1766-11ef-8972-0242ac110002
process[rosout-1]: started with pid [149665]
started core service [/rosout]
process[demo_server-2]: started with pid [149668]
process[service_client-3]: started with pid [149673]
[ INFO] [1716291516.825454846]: waitForService: Service [/yolo_service] has not been advertised, waiting...
process[nodelet_manager_pc-4]: started with pid [149693]
[ INFO] [1716291516.900125932]: Initializing nodelet with 20 worker threads.
process[EstimatorNodelet-5]: started with pid [149708]
[ INFO] [1716291516.970929060]: Loading nodelet /EstimatorNodelet of type vins_estimator/EstimatorNodelet to manager nodelet_manager_pc with the following remappings:
[ INFO] [1716291516.971241555]: /camera/color/image_raw -> /d400/color/image_raw
[ERROR] [1716291517.003830591]: Failed to load nodelet [/EstimatorNodelet] of type [vins_estimator/EstimatorNodelet] even after refreshing the cache: MultiLibraryClassLoader: Could not create object of class type estimator_nodelet_ns::EstimatorNodelet as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1716291517.003853873]: The error before refreshing the cache was: Failed to load library /root/catkin_ws/devel/lib//libestimator_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /root/catkin_ws/devel/lib//libestimator_nodelet.so: undefined symbol: _Z11DEPTH_TOPICB5cxx11)
[FATAL] [1716291517.003943700]: Failed to load nodelet '/EstimatorNodelet` of type `vins_estimator/EstimatorNodelet` to manager `nodelet_manager_pc'
[EstimatorNodelet-5] process has died [pid 149708, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load vins_estimator/EstimatorNodelet nodelet_manager_pc /camera/color/image_raw:=/d400/color/image_raw __name:=EstimatorNodelet __log:=/root/.ros/log/a90bea7e-1766-11ef-8972-0242ac110002/EstimatorNodelet-5.log].
log file: /root/.ros/log/a90bea7e-1766-11ef-8972-0242ac110002/EstimatorNodelet-5*.log
YOLOv5 🚀 e310fa9 torch 1.10.2+cu113 CUDA:0 (NVIDIA GeForce RTX 3060 Ti, 7965MiB)

```

有error报错
[image]

jianhengLiu, 2024-05-22:
参考 issue: https://github.com/HITSZ-NRSL/Dynamic-VINS/issues/1 和 https://github.com/HITSZ-NRSL/Dynamic-VINS/issues/17

tulingcheng86, 2024-05-22:
> 参考 issue: #1 和 #17

我的opencv版本就是melodic自带的好像，3.2.0,,我卸载了再重装的也是3.2.0。还是报这个错误。
这个项目的依赖我是在我自己已经装好vinsfusion的环境下在下载编译的，cere的版本没有按照https://github.com/HITSZ-NRSL/Dynamic-VINS/blob/main/doc/INSTALL.md这个一样，而是1.14.0版本，其他步骤一致

tulingcheng86, 2024-05-22:
我尝试按照依赖自己建了一个docker运行，能成功运行。。我得排查下之前为什么不行的原因才行

tulingcheng86, 2024-05-22:
> 好的，欢迎分享排查出的问题，谢谢
> […](#)

好的，作者你有关于这个项目的所发表的论文吗，我想学习下。

tulingcheng86, 2024-05-22:
> > 好的，欢迎分享排查出的问题，谢谢
> > […](#)
> 
> 好的，作者你有关于这个项目的所发表的论文吗，我想学习下。

找到了，不需要了。

TAKAGNU, 2024-07-10:
> 我尝试按照依赖自己建了一个docker运行，能成功运行。。我得排查下之前为什么不行的原因才行

您好，我遇到了和您一样的问题，请问您排查出来报错的原因了吗?

tulingcheng86, 2024-07-11:
> > 我尝试按照依赖自己建了一个docker运行，能成功运行。。我得排查下之前为什么不行的原因才行
> 
> 您好，我遇到了和您一样的问题，请问您排查出来报错的原因了吗?

没有。。docker跑起来之后就没管了

closed 2024-11-12 by jianhengLiu